MarksMakerSpace/filament-profiles #102: 🐛 [BUG] Unable to Edit Filament Properties on 3D Filament Profiles

luckygreen, 2025-02-12:
**Describe the bug**
Users are unable to edit the properties of existing filaments on the 3D Filament Profiles website. For example, the filament entry at [https://3dfilamentprofiles.com/filaments/3977/details](https://3dfilamentprofiles.com/filaments/3977/details) contains numerous inaccuracies that do not align with the manufacturer's published print profiles or user experiences. There is no available option to correct or update this information.

**To Reproduce**
Steps to reproduce the behavior:
1. Navigate to [https://3dfilamentprofiles.com/filaments/3977/details](https://3dfilamentprofiles.com/filaments/3977/details).
2. Attempt to locate an option to edit or suggest changes to the filament's properties.
3. Observe that no such option is available.

**Expected behavior**
An option should be available for users to edit or suggest corrections to existing filament entries to ensure information accuracy and reflect manufacturer data and user experiences.

**Screenshots**
N/A

**Desktop (please complete the following information):**
- OS: Windows 11
- Browser: Google Chrome
- Version: 88.0.4324.150

**Smartphone (please complete the following information):**
- Device: N/A
- OS: N/A
- Browser: N/A
- Version: N/A

**Additional context**
The inability to edit filament properties limits the usefulness of the database, as users cannot contribute corrections or updates based on their experiences or updated manufacturer data. Implementing an edit feature or a suggestion mechanism would enhance the platform's accuracy and reliability.

dog-rgb, 2025-02-12:
There is a pen icon behind the title to edit the filament. Is that not visible for you?

markcarroll, 2025-02-12:
Just t tested this and it is working correctly. I also took a look at the values in there for that filament and it matches both their web site and the profiles they list on <https://github.com/TINMORRY/TINMORRY-filament-profile-for-Bambu-printers> 

The edit button is at the end of the filament title:

[image]

closed 2025-02-12 by markcarroll