MarksMakerSpace/filament-profiles #108: 💡 [FEAT] add URL in My Spools table

ch33tah0, 2025-02-21:
Can we get links in the "My Spools" page, the same way it is in "Filament List"?
I find it incredibly helpful to quickly open some colors in a new tab with a middle mouse click (also without reloading the whole "My Spools"-page).

Left My Spools | right Filament List:
[image]

Filament List:
[image]

markcarroll, 2025-02-22:
For you, absolutely.  I’ll add it tonight.

markcarroll, 2025-02-22:
Scratch that, I've just done it and deployed to production. It is now live. Thanks for the suggestion.

closed 2025-02-22 by markcarroll