MarksMakerSpace/filament-profiles #110: Portrait labels

markcarroll, 2025-02-25:
https://www.reddit.com/r/3dfilamentprofiles/comments/1ixai0o/feature_request_portrait_labels/

> I think I want to hang the swatches from their little handles. Would be really nice if I could generate the labels in a portrait orientation so I don't have to look at them sideways forever :D

BradCOnGitHub, 2025-02-26:
Just noticed today that you setup the details page to just print as the label. Seems like it would be fairly easy to remake that with CSS instead and get something powerful and simple.

I might play with Stylus/Stylish a bit and see if I can do it myself from the outside, but if you get to where you want to share or publish code let me know.

markcarroll, 2025-02-28:
Hey @BradCOnGitHub can you elaborate on what you are looking for? I'd like to help.

BradCOnGitHub, 2025-02-28:
To be honest, I'm not sure I know yet. So I'm just seeing if I can customize and then iterate to find what works for me.

I had been using a customized version of these: https://makerworld.com/en/models/395861-filament-dry-box-hinged-label#profileId-297168

[image]

They have HUGE text that I can easily read from across the room. Now that I'm trying something different I'm discovering what's important to see from far away and it's really just material and type. The brand is secondary since I rarely have 2 brands in the same color/material but a small version of that would sometimes be nice and will at least make managing the swatches themselves easier over time. I can see the color itself so that text isn't so important either. Print settings and such are never going to be typed in from a label, so just the QR code to get to the page and I can see it there. 

So a rough outline of what I actually NEED would be like this: 

[image]

Maybe add the filament ID in there, but I see a lot of them have it has part of the color name anyway.

Amoenus, 2025-03-04:
To piggy back on this idea, just general "customizable" labels would be nice. Selfishly I would love some smaller templates for my NIIMBOT D110 label maker that accepts 30x12mm and 40x12mm labels, so longer but narrower

[image]

markcarroll, 2025-11-09:
Th problem I've found with each of the other label sizes that I've tried is that it is really hard to get a template that works for all the different lengths of text in each field.  I like the 30x12 that you have above @Amoenus but that only works for very short brand and color names.  I would really like to offer more sizes but getting a generic template that works is harder than it looks.