MarksMakerSpace/filament-profiles #112: Scan Import Info

mharris1984, 2025-03-11:
First, thank you for this amazing project!  I added feature requests many months ago and you added them within a day or two, lol.  But your work is great and I've been using this to slowly log in all my filament and now.  I have a good collection, so its taking me some time.  

I dont have the lables you offer pre-made, but use a Dymo label printer and the 1x2 1/8 labels to log my info with a QR code on it.  It would be nice If you offered multiple label types or sizes  for the info.  But thats something I can live with, haha.  

My feature request is to be able to scan the barcode on the box of filiment to add to my inventory.  So when I get an order of filiments, all I have to do is use my USB barcode scanner (or cell phone camera for some really advanced input on the mobile web version) and it auto adds that roll to my list.  USB scanner is basically just a HID device that types in the SKU and presses enter.  So that would be easy if you have the SKUs for all in the database.  So a section under my filiments for "Scan and add" and just a search box that I scan my SKU or enter a SKU and push enter and it auto adds without any prompts, just a message at the top saying "BambuLabs PLA Basic BLUE added to your filiments" or something.

mharris1984, 2025-03-11:
My last feature request was in Nov #22 , lol.  Seems the site is getting much busier now!  Unfortunitely none of my friends are into 3D printing, so I havent been able to spread the word, ouside of Reddit posts, 👍 I ran accross it just searching for something that makes tracking my filimant on hand easier.  

I had been using an excel spreadsheet that auto generated the QR code for me with all my info needs that I filled out in the cells.  But is was a pain to keep up. 

Also the new .3MF upload to auto subtract the ammount out of your filimant is amazingly awesome!!!!   I'm going to start scanning all my prints when I start a new roll of those colors at 100. so my inventory will start to catch up

markcarroll, 2025-03-11:
ok this I am definitely interested in doing.  so if we add the SKU as a searchable field the scanner will just type it in?  This sounds like a brilliant idea.  What scanner are you using? I am just about to release an upgrade to the input box, one feature of which will be to add a SKU field.  We are getting close.

markcarroll, 2025-03-11:
oh and share away on Reddit and anywhere else... 😁

markcarroll, 2025-03-11:
ok, I've ordered a scanner.  This will be fun.

mharris1984, 2025-03-13:
Sorry, I just saw the replies!  

The scanner I'm using is just a normal USB barcode scanner.  Basically what is does, is acts like an HID device (Human Interface Device - think keyboard/mouse) so what it does is type in the barcode numbers for me and simulates pushing ENTER on the end.  

So if I'm in an Excell Spreedsheet and I scan a barcode on a box from my BambuLabs order, it will put the SKU in the cell and auto go to the next cell, as if I pushed enter key.  

So the idea is a bulk add area in "My Spools" to scan a rolls barcode and when you scan it, it auto pushes ENTER or you can push a button next to it that does the same thing for "Add Spool" so if a scanner doesnt do the pressing enter automattically.  And so if I scan a roll of BLACK PETG-HF bambuLabs, the SKU scans and auto adds a spool to My Spools.  Makes adding new spools easy!  Where right now, i'll order 20 rolls of all diffrent colors and such and so i have to search and add manually.  Having 129 spools on hand of all diffrent colors/meterials is a pain to manage that way, lol.  

Maybe have a roll ammount next to it?  so if I get 20 rolls of BLACK, I can put 20 in the qty box and then scan the barcode and poof, all there.  But have it default to one.  If I have 2 or three rolls of the same kind in my order, I dont mind just scanning each barcode and have them all inserted in about 30 seconds.  But if I have a bulk order, it would be nice to set the nubmer and scan once and not have to scan all 20 of the same color :)

I hope this makes sense :)

mharris1984, 2025-03-21:
Sweet!!  That is awesome!!!  If you need anyone to test or anything, feel free to reach out!

splatski, 2025-08-07:
I have the same sort of desire to be using my barcode scanner. I found that of the things I found on the site with a barcode doesn't match the barcode on the box of that filament I have. so I'm unsure if there could be some regional differences or something but most of my filament doesn't have barcodes on the site and I don't see any way to submit the barcodes.

markcarroll, 2025-08-14:
have you tried scanning sku/upc codes into the search box? This works now if the SKU/UPC exists.

markcarroll, 2025-08-15:
Yes I've thought about a bulk import.  I just released location scanning so this seems like a good follow up.