MarksMakerSpace/filament-profiles #118: Can't change PA to PPA to fix Bambu PPA-CF entry

PPFilip, 2025-04-01:
Tried to add some details to previously created entry by someone else here https://3dfilamentprofiles.com/filament/details/4110

This PPA-CF profile has material as PA and material type used to be "Metal" which is not correct.  It looks like I can't change PA to PPA thought. I am not sure if this ignore of material change is a bug or a feature though.

markcarroll, 2025-04-03:
It looks like someone has added a PPA version here https://3dfilamentprofiles.com/filament/details/5661

If that is correct I will redirect the old one ([4110](https://3dfilamentprofiles.com/filament/details/4110)) to [5561](https://3dfilamentprofiles.com/filament/details/5661). 

In answer to your question, I had to remove the ability to edit filament types once they had been added to someone's My Spools as we had some people abusing the system and completely changing the filaments that folks had in their spools list.

PPFilip, 2025-04-03:
Yes, they seem identical. Thank you for explaining as well

markcarroll, 2025-04-05:
Thanks for raising the issue.

closed 2025-04-05 by markcarroll