MarksMakerSpace/filament-profiles #122: 💡 [FEAT] Add direct access QR code for weight

thezemo, 2025-04-13:
**Is your feature request related to a problem? Please describe.**
While the qr code system is very nice to have to add a spool to my collection, after the spool is added there is no easy way to manage the collection without logging into the website.

**Describe the solution you'd like**
Allow for a qr code that would directly access a particular spool in my collection once added, that way I can easily manage weights without having to link directly into Bambu system (manual weighing)

**Describe alternatives you've considered**
I currently use https://www.printables.com/model/498719-filament-tracking-with-google-sheets-qr-codes however with very small tweaks I believe this approach could be used within this program for weight and spool inventory management

**Additional context**
Add any other context or screenshots about the feature request here.

markcarroll, 2025-04-13:
Hi. Thanks for the feedback.  I am about to start working on exactly this.  I think having a QR code per spool, rather than per filament type, that when you scan it a simple form pops up that allows you enter the current weight or change its location, but very simple and mobile focused.

thezemo, 2025-04-14:
The way it works currently is every spool has a QR code. Unfortunately you have to enter the info in the Google Sheets about the roll before you can get the QR code. After you enter the info though, you can scan the  QR code and it will bring you to a Google Form that has the spool number and description pre-filled in (so you can verify the spool i guess) you enter the weight and hit submit, that is it, it is automatically added to the Sheets and the remaining amount is auto updated.

I have added a note field that allows me to add notes likes "New" "Predry" "Afterdry" "Project Name" whatever I want really.

I like that is not forcing intergration with a slicer, but think it would be nice if it could intergrate thus giving you one spot that easy to maintain the inventory and weights and also the print history etc (this is above and beyond though)

Any questions or if you need someone to test out things let me know.

markcarroll, 2025-05-18:
@thezemo I have rolled out a per-spool QR code feature.  By each spool for a color in your My Spools section, there is a QR code button that will give you a code specific to that spool. I plan on evolving this some more in the near future to add QR code scanning into the site.

closed 2025-06-21 by markcarroll