MarksMakerSpace/filament-profiles #125: Fix Polymaker "Marble" types

minsis, 2025-04-18:
**Is your feature request related to a problem? Please describe.**
Polymaker "Marble" type is actually PolyTerra. Seems like maybe they should all match to be PolyTerra type filament with "Marble" in the name?

[image]

markcarroll, 2025-04-18:
From their website it looks like they rebranded PolyTerra Marble as Panchroma Marble. I really don't want to be chasing constant rebrandings. I can bulk update these to the new Panchroma Marble if that is the new name since I just added it for you. 

> Panchroma™ Marble PLA (Formerly PolyTerra™ Marble)

https://us.polymaker.com/products/panchroma-marble?srsltid=AfmBOootzrelBcV43WUhBAGFp4LiDswPS6gcdmvpzAz2XRKj2zSInPQP

markcarroll, 2025-05-17:
@minsis I'll close this if there is nothing else to do here. Re-open it if you want me to do more.

closed 2025-05-17 by markcarroll