MarksMakerSpace/filament-profiles #127: [FEAT] Add 24mm (0.94 inch) Brother P-Touch Swatch Card Label output

cswingler, 2025-04-20:
**Is your feature request related to a problem? Please describe.**
The label output can't directly be printed on my [Brother P-Touch PT700 Printer](https://www.brother-usa.com/products/ptp700). P-Touch printers have limited DPI, but they're cheap and ubiquitous!

A 1.5" long label using 0.94" tape fits nicely on Filament Swatch System cards, but the output definitely needs to be condensed to make them readable and the QR code still scannable

**Describe the solution you'd like**
A Swatch Card Label output that fits on a 0.94" by 1.5" P-Touch label

**Describe alternatives you've considered**
I've manually created them in P-Touch editor, but it's time consuming and error-prone.

[image]

[image]

Taking one of the existing swatch outputs and sending it directly to my printer results in illegible output, and an un-scannable QR code.

[image]

**Additional context**

markcarroll, 2025-04-20:
Thanks for all the detail. I have a work item on my plate to produce a vertical label, so I will use this as a guide.

markcarroll, 2026-03-16:
@cswingler I am working on some new label formats right now.  I am planning on adding some vertical ones and would like to see if I can get one that fits on your printer size.  Is the picture at the top the best you were able to do with that size?