MarksMakerSpace/filament-profiles #131: Custom Reference Number

raulgotor, 2025-05-02:
**Is your feature request related to a problem? Please describe.**
I use my swatches collection to decide what colour/combination of colours  I want to use in my current project. Since those are small 2x2cm swatches, I only have a custom reference number such as KH23 written there, and with that, I go to a list I have, and I check what brand/material/type that KH23 translates to. I started using 3dfilamentprofiles short time ago, and I didn't found such a custom field, therefore I'm doing a work around that is using the Location field for that. It gets listed, but one cannot use the search field for that, so still a manual search involved

**Describe the solution you'd like**
Possibility to add general purpose custom fields that can get listed, but also that can be searched

**Describe alternatives you've considered**
As explained before, I'm doing a work around that is using the Location field for that. It gets listed, but one cannot use the search field for that, so still a manual search involved

**Additional context**
-

markcarroll, 2025-05-02:
There is a notes field for your spools but I am not sure that search checks that either.  I will add the both to the search field list though.

markcarroll, 2025-05-02:
ok I've deployed an update that searches the location field.  Notes are not on the summary page but I'll work on a way to pull them in that makes sense.

closed 2025-05-02 by markcarroll

raulgotor, 2025-05-03:
Wow @markcarroll, that was quick! Thank you very much. Also, let me know if there is some ticket you'd like some help with