MarksMakerSpace/filament-profiles #134: 💡 [FEAT] API

DocJ-GA, 2025-05-17:
I am working on a console app to sync spoolman filaments to orcaslicer (https://github.com/DocJ-GA/Whaler).

I would like to have an api with 3dfilamentprofiles to grab the awesome information you have and either put it into spoolman instances via plugin.  Any thought on letting this happen?

andrewash, 2025-05-23:
This would be amazing. Spoolman's native app's UX is pretty good, however I found the data had lots of community-contributed duplicate entries for the same filament and plenty of gaps.

With recent AI driven tools it would be wonderful to see a new renaissance of open APIs for sites like 3dfilamentprofiles - who knows what 3D printer folks could create with that API?

Keeping the quality of data high is important. Even if it was just read-only access to the database, plus write access to "My Spools" that would be useful.

jdh313, 2025-05-24:
I'd be interested in this too! However, I'd be worried about putting in significant effort without getting clarity on this in the ToS:

> No Scraping or Automated Tools – You may not create, use, or distribute any tools or scripts intended to scrape, copy, or clone the database in whole or in part.

markcarroll, 2025-05-25:
I am open to the idea of integrating with other tools and offering APIs for certain functions, however I am not clear yet on the right way to do this without incurring more cost to me for web and database usage.  I have to pay for both as usage exceeds their free plans. In fact, with Vercel we even exceed the amount they give you in the Pro plan and I have to pay incremental costs on that.  So, without a clear strategy for how to limit API usage, or recoup costs through said usage, I am holding off on opening up APIs for now.   

I am open to suggestions here though.  I have built platforms for most of my career so would love to open things up here as well.  I just haven't come up with the right approach yet, nor had the time to think it through.

andrewash, 2025-05-25:
@jdh313 AWS or DigitalOcean is probably more affordable. 

But is this project centered in Europe and avoiding hosting data in the U.S.?

A freemium iPhone app model is one option. Like "scan 25 rolls of filament for free to quickly add them to My Spools. Then for an $X In App purchase they can scan unlimited spools. Proceeds after Apple's cut would go towards your hosting costs. 

A more basic option is just charging for a certain number of daily or monthly API Requests. I'd likely pay just to be able to build an app for my personal use. But that doesn't help the whole community .

an0Nym0us63, 2025-09-16:
i totally +1 this demand,  would pay a fee without hesitation to integrate your service in my personnal spool tracking tool

escidmore, 2026-03-03:
I would also pay for API access, if it was available.

Crash0Labs, 2026-09-26:
+1 for this — adding a second use case that extends the original request and I also would pay for a preimum sub.

Use case:
I run Bambuddy (self-hosted Bambu Lab printer manager), which already has its own REST API with spool inventory endpoints (https://wiki.bambuddy.cool/features/api-keys/). I'd like to keep My Spools in sync with it automatically — remaining weight after prints, new spools added, storage location — without manual CSV export/import.

This goes a step beyond the catalog read access requested above: it would need per-user API tokens so a tool can read and update *my* spools.

Suggested scope (read-only first, write later is fine):
- Personal API token generated in account settings (revocable; ideally read-only vs read/write scopes)
- GET my spools (with a stable spool ID, weight, remaining, location, filament ID)
- GET filament details by ID (catalog lookup — this also covers the original request here)
- PATCH a spool (update remaining weight / location)
- POST a new spool

To make two-way sync possible later (not just one-way):
- A last-modified timestamp on each spool, plus a way to fetch only spools changed since a given time (e.g. GET /spools?updated_since=...)
- An optional external_id field on spools, so a sync tool can store the matching Bambuddy/Spoolman ID and avoid duplicates
- Webhooks on spool changes (nice-to-have; polling with updated_since works fine otherwise)

Keeping server cost down:
- Per-token rate limits would be completely reasonable — a sync tool only needs to check every so often, not constantly.
- With updated_since, each check returns only what changed, so most requests would be tiny.
- Starting read-only keeps both the load and the security surface small.

Related: #628 — as noted there, Bambuddy's CSV export has no spool ID, so two-way sync isn't possible via CSV. An API exposing stable spool IDs would solve that. Also #659 (updating existing spools) and #311 (Spoolman integration).

With this in place, Bambuddy, Spoolman, Home Assistant, or a small sync script could all integrate without you needing to build and maintain each integration yourself.

Thanks for all the work on the site!

litebito, 2026-09-27:
> +1 for this — adding a second use case that extends the original request and I also would pay for a preimum sub.
>
> Use case: I run Bambuddy (self-hosted Bambu Lab printer manager), which already has [...]

Exactly my use case also, +1

markcarroll, 2026-09-28:
> Depending how big your database is, I would be happy to help by load balancing the traffic to my shared hosting which isn't using half of its full capacity anymore.. I've got 20+ GB of space to spare... Bandwidth: 351.5 MB / Unlimited Total Disk Usage: 20.64 GB / 48.83 GB Number of MySQL DBs: 10 / 25
@t4ai-workshops i appreciate the offer but my data volumes have grown immensely and, for example, some months I server up well over 1TB.   I have been working on better caching mechanisms but it has ended up quite complex.  

@Crash0Labs @litebito i am traveling right now but I will look through your request when I am back and see what I can do.