MarksMakerSpace/filament-profiles #144: Make website code open source

queengooborg, 2025-06-29:
I was surprised to see that this project's source code had not been made public, and I would love to see the source code for the website made open source.

A lot of the features that I immediately desired when first starting to use the website, I noticed that others had also requested.  So, my first inclination was to take a look at the source to see if it was something I could implement and submit as a PR to be merged upstream.  When I wasn't able to find the source repository so I could do just that, I was surprised, and also a little disappointed.

I feel that making the site code open source would help to reduce maintenance burdens by allowing additional developers outside of the core team to contribute, and add to the collaborative, community effort experience that drives the whole project.

Snaxmuppet, 2025-07-08:
This would be a good option for a project like this. The owner could maintain control of the master but we would then be able to code up suggestions and fix bugs without it taking up time for the owner.

litebito, 2025-12-28:
I also would be interested in running this on my home network