MarksMakerSpace/filament-profiles #148: 💡 [FEAT] User set defaults for Swatch Card Labels - faster printing options

baulchjt, 2025-07-10:
**Is your feature request related to a problem? Please describe.**
For faster printing of Swatch Card Labels without needing to make adjustments to each label I would like to print  (eg, selecting PNG, slots 1-4 preferences)- allowing for the user to set default preferences on their profiles, so anytime the want to do this action, it is saved already as a custom user default.

**Describe the solution you'd like**
It would be great if some defaults could be set based on my user profile, for example, when I go to print Swatch Card Labels, and I always need to use the PNG version, that I could set my profile default preference once, so that anytime I select the image on any spool, it will download the PNG as my preferred default. As well as my preferred defaults for Slots 1 > 4.  

**Describe alternatives you've considered**
Alternatively, maybe adding a "print" icon on the Swatch Card Label pop up from the Action Buttons (in the list view), so rather than downloading a file then printing, by default if you click on the print icon, it sends a generic print file (PNG, PDF, or SVG) that most label printers can read.  Unless I am missing some basic functionality that already does this and I totally missed it. . . :S 

**Additional context**
You provide a great product, and I really appreciate it!  This is just a small annoyance that I am looking to reduce a few clicks :)  Thank you!

logikal, 2025-07-23:
@baulchjt I'm not affiliated with the project, but I think this got implemented!

If I change the type and fields, then close and go to a different filament or spool QR, I see the behavior you're looking for.

@markcarroll must've slipped this in!

baulchjt, 2025-07-23:
@logikal - Thanks, I see it too!

@markcarroll  - Thank you!  Much appreciated!

closed 2025-07-23 by baulchjt

markcarroll, 2025-07-23:
Thank you @logikal for updating this ticket.  Yes, @baulchjt I did indeed slip this one in a few days ago. Sorry I forgot to update you.  It does only save your preview  settings to your local storage but at some point I will move that to be tied to your account.