MarksMakerSpace/filament-profiles #154: Add "Location" to labels.

fireymerlin, 2025-07-20:
Several places in the online "Help" state that "location" can be placed on the label but it is not available in the drop-down fields.

markcarroll, 2025-07-23:
ok I think there may have been some confusion here. The idea is that you can print a label for a specific spool. The QR code on that label will take you to a page just for that spool, where you can update the current weight and location of that spool quickly and easily.   Having the location on the label that is stuck on the spool doesnt make a lot of sense to me.  

Are you using it differently?

fireymerlin, 2025-08-05:
> Mark - Wouldn't one use case for the location on the label be for
> returning spools from my AMS to their 'proper' storage location (without
> having to look up the location in the database)?

I put a small label on the spool with the location so I know where to
return the spool.  Having the location on the label generated by the
website makes a lot of sense to me.  It would save me from using 2 labels
per spool.  I have over 100 spools, over 20 dry boxes.

closed 2025-08-05 by markcarroll