MarksMakerSpace/filament-profiles #156: Embed spool ID in Label and label filename when printing spool labels

logikal, 2025-07-23:
**Is your feature request related to a problem? Please describe.**
I have 4 spools of the same filament, and often order multiple spools at a time that I'd like to have in 3dfilamentprofiles for inventory management. When I downlad the labels to batch print, I just get my operating system numbering for duplicates.

**Describe the solution you'd like**
Include the spool ID in the filename and label. Allow it to be shown as a column in the spool list for a filament. This would allow me to make sure I'm applying the right label to the right spool without having to read the QR code.

**Describe alternatives you've considered**
Struggling to come up with alternatives.

**Additional context**
I might be the weird one here that has multiple spools of the same filament open at a given time - if so, I totally get that this feature might not be useful enough to the broader user base, and I'll manage 😃 .

logikal, 2025-07-23:
Awesome, thank you!

markcarroll, 2025-07-26:
[image]

[image]

closed 2025-07-26 by markcarroll