MarksMakerSpace/filament-profiles #157: Spool weight inconsistent between mobile and desktop

logikal, 2025-07-25:
**Describe the bug**
On desktop, when editing a spool, the spool weight can't be changed and is set to 250g.

On mobile, spool weight can't be edited and defaults to 0.

Since I run a lot of different spools, I prefer the editable 0 option.

[image]

markcarroll, 2025-07-25:
Ooh that’s a weird one.  Ok so you want the preset value to appear in there if we have one but have it still be editable so you can correct it if needed.  That makes sense.   I’ll take a look when I get a chance.

logikal, 2025-07-25:
It looks like all the spools  (plastic, cardboard, refill, other) are uneditable, and set to 250g. If there was a way to say “Bambu HT spool” and that was always 260g, or “overture cardboard” and that was 147, that’d be baller!

markcarroll, 2025-07-25:
it should be reading the empty spool weight from the filament settings.

markcarroll, 2025-07-25:
@logikal can you confirm whether the screenshots you showed were from the same filament? if not then I think the differences you are seeing are that the different filaments have different empty spool settings.

logikal, 2025-07-25:
> [@logikal](https://github.com/logikal) can you confirm whether the screenshots you showed were from the same filament? if not then I think the differences you are seeing are that the different filaments have different empty spool settings.

It was the same specific spool, not just filament

markcarroll, 2025-07-25:
ok that's really weird then.  I am not able to reproduce it.  I did just deploy an update that removes the read-only flag for preset values. Now it just puts up a message but you can override it.

logikal, 2025-07-25:
Now I just gotta go edit the default spool weights of all these filaments 🫡