MarksMakerSpace/filament-profiles #160: My Spools - Notes

fireymerlin, 2025-07-27:
**Describe the bug**
If I'm on the "My Spools" window and click on the edit button, the Notes field is not being populated.  I can put notes in and they will show on the other Filament details pages but not on the one accessed by clicking edit on My Spools.

**Screenshots**
This "Edit Spool Details" was accessed from the "My Spools" edit button.

[image]

This "Edit Spool Details" was accessed by using "My Colors", clicking on the spool, clicking on the edit button.

[image]

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

markcarroll, 2025-07-27:
thanks for pointing this out. it is actually a little tricky as the "My Spools" page does not have the notes available.  I will work on getting this fixed.  The notes are there, they are just not available to that page.   Stay tuned.

markcarroll, 2025-08-02:
This should be fixed now. Please re-open if it is not working correctly for you.

closed 2025-08-02 by markcarroll