MarksMakerSpace/filament-profiles #165: Spool label preview doesn't match downloaded .aml file

logikal, 2025-08-12:
**Describe the bug**
Spool (not filament) label preview doesn't match downloaded .aml file. 
 
**To Reproduce**
Steps to reproduce the behavior:
1. Go to my spools
2. Click on a filament you have
3. Click the SPOOL QR.
4. Compare the preview to the downloaded file.

**Expected behavior**
I expected the QR codes to match - previously the circle in the top left has been a spool QR code, and a rectangle in the top left has been a filament QR code. 

**Screenshots**

[image]

https://github.com/user-attachments/assets/661b048a-fa2b-4af8-a9a2-bafca462c5e7

**Desktop (please complete the following information):**
 - OS: MacOS
 - Browser Arc (Chrome Based)

**Additional context**
If this is just a change that's intended (i.e. spool is denoted by circle in the center, I'd expect the preview to match. 

Thanks!

markcarroll, 2025-08-12:
yeah that is not right.  I am in the middle of re-doing the labels anyway so I will take a look at this today. Thanks for reporting and for taking the time to do a screen recording.

logikal, 2025-08-12:
> yeah that is not right. I am in the middle of re-doing the labels anyway so I will take a look at this today. Thanks for reporting and for taking the time to do a screen recording.

of course! thanks for the app, and for being so responsive!