MarksMakerSpace/filament-profiles #171: SVG labels are not usable.  PNG are.

fireymerlin, 2025-08-16:
**Describe the bug**
I have been creating SVG labels and inserting them into a MS Word document with an Avery type of label.
Today, 8/16/2025, I get a message in word that "The picture can't be displayed.".  I also can't open the the .SVG file in MS Edge, I get an error.
SVG files I created a day or two ago still work.

markcarroll, 2025-08-16:
I just created a Windows VM so I’ll give this a try.

ch33tah0, 2025-08-17:
the error in chrome:
[image]

works again, after you add this to the svg tag **xmlns:xlink="http://www.w3.org/1999/xlink"** in chrome and MS Word.

<svg viewBox="0 0 400 300" xmlns="http://www.w3.org/2000/svg" **xmlns:xlink="http://www.w3.org/1999/xlink"** id="label-svg" class="mx-auto w-full max-w-[400px] rounded-2xl border border-gray-300 shadow-md print:mx-0 print:block! print:border-none">

markcarroll, 2025-08-17:
Thanks for the extra info. I will try and fix this later today.

markcarroll, 2025-08-17:
thanks for the extra info.  I just pushed a fix. take a look and let me know.

fireymerlin, 2025-08-17:
Looks like it is working now.

closed 2025-08-17 by markcarroll