MarksMakerSpace/filament-profiles #208: Hide empty spools

SiDtheTurtle, 2025-09-22:
Love the feature that shows if you have zero spools of a colour in red. Is it possible to hide them, either globally as a setting or better per filament or per location?

I keep my used spools to track how much I'm using, so as they run out I set them to 0g, and have them in a specific location I called 'Spent'. However, there are cases where I no longer want to use a colour, because I'm using an alternative. For example I'm switching my greys out with a nicer matte grey. So now, the old basic grey PLA that I did buy loads of, but won't any more, I don't need that in red telling me I've run out.

Doing it globally as a setting may be too much of a hammer to crack a nut, but perhaps I could flag a filament as 'do not show when empty', or perhaps by location, you could have a magic location that's invisible outside of the location screen.

SiDtheTurtle, 2025-09-29:
Cheers Mark. I like to keep a history of everything I've used, helps with tracking prices historically, working out my total spend and usages, so I don't want to remote them from the system.

Perhaps combining it with the wishlist feature, I'm done with these greys so I'd flag them as do not show once they run out, but otherwise continue flagging other colours as red and perhaps summarise all empty colours except those that were flagged in a shopping list screen?

hansenc0705, 2025-10-07:
Was happy to see them turn red, makes it easy. I'm also moving those spools to a custom "reorder" location so they are in one place.

markcarroll, 2025-10-30:
I have a separate ticket for adding wishlists, so I think I will close this one.